openhwfoundation/core-v-xif #30: Increase minimum ID_WIDTH

Silabs-ArjanB, 2021-11-30:
ID_WIDTH is currently specified as being between 1 and 32. Requiring support for ID_WIDTH = 1 causes artificial corner cases already in relatively short pipelines. E.g. a CPU would have to put back pressure on its compressed interface if there is already more than 1 other offloaded instruction in flight (which would be quite normal). Mandating a minimum width of say 4 would allow for 16 in-flight instructions, which would not require a back pressure mechanism to be implemented on normal use cases for the current OpenHW cores that are on the roadmap. On the CPU side this does not imply additional hardware cost as the CPU is still free to only support only 1 or 2 in-flight offloaded instructions (and tie of some MSBs of the ID). On the coprocessor side the increased minimum does imply an increased minimum cost (but not added complexity).

davideschiavone, 2021-12-02:
Hi @Silabs-ArjanB, as it is a PARAMETER (1...32), one can simply recommend using a minimum of 3 or set 3 as a Default parameter without forcing other implementations to have 3bits and then playing with the MSB to pretend it's smaller.
I suggest we keep 1...32 and set as default and recommended 3

Silabs-ArjanB, 2021-12-02:
Hi @davideschiavone If we stick to 1..32 then we mandate that all CPUs support these settings, which implies complications. As soon as we start allowing a subset of parameters to be supported, then we are lost with many mutually incompatible processors and coprocessors.

davideschiavone, 2021-12-03:
I see you are right - merged

closed 2021-12-03 by davideschiavone