openhwfoundation/core-v-xif #36: XIF needs access to mstatus

Silabs-ArjanB, 2022-01-17:
The vector spec states:

- Attempts to execute any vector instruction, or to access the vector CSRs, raise an illegal-instruction exception when mstatus.VS is set to Off.
- When mstatus.VS is set to Initial or Clean, executing any instruction that changes vector state, including the vector CSRs, will change mstatus.VS to Dirty. 

Therefore mstatus needs to become part of the XIF protocol (e.g. part of the issue/result interface)

A similar requirement exists for the FPU

michael-platzer, 2022-01-19:
The new extension status fields are also part of the newest version of [the RISC-V privileged spec](https://github.com/riscv/riscv-isa-manual/releases/download/Priv-v1.12/riscv-privileged-20211203.pdf), where they are discussed in Sect. 3.1.6.6. Essentially, the `mstatus` CSR has 3 extension context status fields named `FS`, `VS`, and `XS` that contain the context status of a floating-point unit, a vector unit, and other potential extensions that need to save/restore state, respectively.

These extension context status fields are 2 bit wide each and their content corresponds to the values `Off`, `Initial`, `Clean`, or `Dirty`. Implementations are allowed to track the dirtiness of each field imprecisely, by setting it to `Dirty` whenever there is an attempt to write the value `Initial` or `Clean` to it. Thereby the context status can actually be stored in 1 bit, which is either `Off` or `Dirty`.

As far as I can tell, the other fields in `mstatus` are not relevant to a coprocessor (or are already part of the XIF interface, such as the privilege mode).

In the context of the XIF interface, the coprocessor needs to know whether these fields are set to `Off` or not (and reject the instruction in case the required field is `Off`). Hence, this would have to be part of the issue interface (and probably also part of the compressed interface as well, since that is also used to signal whether an instruction is accepted).

The main core needs to know when any of these fields needs to be set to `Dirty`. The coprocessor could already signal whether an instruction causes any of the fields to become dirty via the issue response, and the main core could then overwrite the field once the instruction retires, or the coprocessor could signal this via the result interface. If the dirtiness is tracked imprecisely by simply setting the context status field to `Dirty` right away, then this would not actually need to be implemented, since the field is already dirty if the extension is enabled.

davideschiavone, 2022-01-20:
@moimfeld, how are you dealing with it?

moimfeld, 2022-01-28:
Hi,
Since my subsystem is tailored to cv32e40p and the `FS` status field of `mstatus` is not implemented in cv32e40p (see [documentation](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/control_status_registers.html?highlight=mstatus#machine-status-mstatus)) I did not take it into account when designing the FPU subsystem, so I am not dealing with it at all.

However, while reading the specification I came to the conclusion that it needs to be part of the x-interface. I suggest exposing these registers to the coprocessor via the issue interface as well as the compressed interface. I also agree with Michael that you could tell the core via the result-interface to change the value of the register.

The only issue I see is that we are adding extension-specific signals to the interface. I am not very familiar with `mstatus`, but it looks like custom extensions cannot have status fields like `FS`, `VS` and `XS`. So it might not be such a big problem after all, because the interface would still be generic even though it contains these extension-specific signals.

michael-platzer, 2022-01-28:
@moimfeld After reading the relevant section of the spec again, and If I understand the specification correctly, custom extensions cannot have a status field in `mstatus` but can define one in a custom CSR. The `XS` field is a read-only field in `mstatus` that summarizes the status of all user-defined extensions. If the `XS` field is set to `Clean`, then none of the custom extensions has a `Dirty` state and thus a context switch routine does not need to check the status of all custom extensions.

Since the `XS` field is read-only and summarizes status fields implemented in custom CSRs, it does not need to be part of the XIF issue interface (each custom extension would check its own status field in a custom CSR to see whether it is currently enabled). However, the coprocessor would then need to be able to update this field (either as a result of an offloaded CSR instruction that has modified a status field in a custom CSR, or as a result of an offloaded instruction that modified that state of a custom extension).

The `FS` and `VS` status fields in `mstatus` are specific to the F and V extensions, respectively. These are `WARL` fields in `mstatus`, so their content must be communicated to the coprocessor in the XIF issue and compressed interfaces, as you said, and updated in case an offloaded F or V instruction modified the coprocessor state.

Hence, although these fields are extension-specific, adding them to the XIF interface is probably as generic as the `mstatus` CSR itself.

moimfeld, 2022-01-31:
@michael-platzer I agree with you that the possible issue I mentioned is not a problem then.

Silabs-ArjanB, 2022-02-01:
@michael-platzer I will do a pull request today with a proposed way of implementing the above. The only thing that seems awkward to me is that XS represents the 'combined' state of all of the additional user mode extensions. Of course XIF is a point to point interface, but if we use XIF to build connect a bunch of coprocessors, then somewhere the state of all coprocessors need to be combined into this combined XS state. I propose that this is the responsibility of the interconnect (so out of the scope of the XIF spec itself).

Silabs-ArjanB, 2022-02-01:
Hi @michael-platzer , @moimfeld Can you please have a look at https://github.com/openhwgroup/core-v-xif/pull/42 to see if you agree with that as a solution for this issue?

closed 2022-02-10 by Silabs-ArjanB