openhwfoundation/core-v-xif #41: Allow combinatorial path inside processor from memory request interface to memory response interface

Silabs-ArjanB, 2022-01-19:
The memory response (mem_resp) will depend on the memory request (except for simple processors that cannot cause a synchronous exception or 'before timing' debug trigger match based on the memory request. Therefore the processor cannot provide a default memory response and if we forbid a combinatorial path, then the memory response can be sent earliest in the next cycle. Therefore the memory request transaction will need to be 1 for at least two cycles per memory request which limits the load/store bandwidth.

I will make a PR to allow such combinatorial paths and to disallow a combinatorial path from memory response to memory request within the coprocessor (to avoid loops).

Silabs-ArjanB, 2022-01-19:
Note: A combinatorial path from result_vvalid to result_ready would not need to be allowed as no data is communicated back to the coprocessor for that interface.

davideschiavone, 2022-01-20:
@moimfeld to check for cv32e40p branch with FPU

closed 2022-02-01 by Silabs-ArjanB