openhwfoundation/core-v-xif #51: The standard needs to carify CPU or coprocessor should kill instructions after an exception

bertwhl, 2022-04-14:
Hello,
In the interface, there are three types of exceptions that both CPU and the co-processor knows, since these exception signals are passed through the interface. The three types are:
- Co-processor raises an exception, and informs it to CPU via result interface.
- CPU raises an exception while using load & store unit, which is dealing with a request from memory interface from co-processor. CPU informs co-processor this exception via 'mem_resp' signal in memory interface.
- Same as the second case, the only different is that CPU informs co-processor this exception via memory result interface.

To support precise exception, after an exception is raised, CPU should kill every instruction after that instruction. For the three types of exception above, the instructions in co-processor that are issued but not committed should all be killed. The question is, since both CPU and accelerator knows this exception, should CPU kill these instructions via commit interface or should the accelerator kill these instructions by itself. I think this should be clarified by the standard.

I think the second one "The accelerator kill these instructions by itself. " is more reasonable. If the CPU kills the instruction via commit interface, it would take multiple cycles because CPU can only kill one instruction at a cycle. And if we choose the second choice, CPU can keep running without waiting for killing instructions.

Silabs-ArjanB, 2022-04-21:
Hi @bertwhl Thank you for the question. The idea was to let the processor be responsible for killing the necessary instructions to ensure precise behavior, but this was indeed not spelled out in the spec. 

Could you please have a look at https://github.com/openhwgroup/core-v-xif/pull/53 to see if you agree with the modified spec?

The reason that the processor should be responsible is that there is an additional type on top of the three types that you describe: Multiple coprocessors might exist and in case one coprocessor sees an exception, then instructions might need to get killed in the other coprocessor(s) as well. Only the CPU has this overview. Also a coprocessor itself does not know which of its instructions are actually to be flushed (it can not differentiate whether the instructions in its pipeline are already part of the exception handler or not).

bertwhl, 2022-04-28:
Yes, I agree with your modification. Thank you very much for the reply!

closed 2022-04-28 by bertwhl