openhwfoundation/core-v-xif #59: Contradiction in handshake dependency rules

michael-platzer, 2022-06-10:
Hi,

The [section about handshake dependencies](https://docs.openhwgroup.org/projects/openhw-group-core-v-xif/en/latest/x_ext.html#handshake-dependencies) states that:
> Transactions with an earlier issued id shall not depend on transactions with a later issued id (e.g. a coprocessor is not allowed to delay generating issue_ready = 1 because it first wants to see result_ready = 1 for an older instruction).

It seems to me that the example contradicts the rule. To my understanding, an earlier issued id would correspond to an older instruction, whereas a later issued id corresponds to a newer instruction. Hence, the rule states that older instructions (with earlier issued id) shall not depend on newer instructions (with later issued id), but the example claims the opposite, i.e., that a newer instruction must not depend on an older one.

I believe that the example's requirement that `issue_ready` must not be delayed due to waiting for `result_ready` of an older instruction cannot be met. As the coprocessor's pipeline fills up it will eventually need to delay accepting further instructions until earlier results are transmitted to the CPU.

Silabs-ArjanB, 2022-06-14:
Hi @michael-platzer , you are right, the example given is wrong. Instead of just changing 'older instruction' into 'newer instruction' I also changed the used signals (as otherwise the example would become impossible).

I am now proposing this:

> Transactions with an earlier issued ``id`` shall not depend on transactions with a later issued ``id`` (e.g. a |coprocessor| is not allowed to delay generating ``mem_valid`` = 1 because it first wants to see ``commit_valid`` = 1 or ``result_ready`` = 1 for a newer instruction).

Does that look okay to you?

Silabs-ArjanB, 2022-06-14:
@michael-platzer Could you please have a look if https://github.com/openhwgroup/core-v-xif/pull/60 fixes this issue?

michael-platzer, 2022-06-14:
Hi @Silabs-ArjanB, thank you for your reply and the pull request! Your changes fix the issue. I believe that the handshake dependency rules are clear now.