openhwfoundation/core-v-xif #67: Parameterization of capabilities

christian-herber-nxp, 2023-05-15:
I have a question on expectations and decision making in the interface. As an example, lets take dual read ability in the issue interface.

The specification will require the interface to include a `logic [2:0] dualread` signal no matter what. There is no parameterization of the signal based on `X_DUALREAD`. So the specification admits that some cores will not support dual read (or just limited), but at the same time wants the dualread signal to be present. Looking at implementations, e.g. CV32E40X does carry the `dualread` signal, while not being dualread capable. 

Can someone give me background why it is done this way?

Silabs-ArjanB, 2023-05-15:
Hi @christian-herber-nxp The major thing is that we do not want any ifdefs in the SystemVerilog code. Given that restriction what would be your proposal?

christian-herber-nxp, 2023-05-15:
for the interface specification, the solution would be to change 

dualread | logic [2:0]

to

dualread | logic [X_DUALREAD-1:0], and stating this only exists for X_DUALREAD  > 0 (i.e. in Table 6).

The system verilog example is very useful, but will require ifdefs to represent the specification above. The specification lists the system verilog as example, not as normative. I think it is fair to state that the example represents a specific configuration.
In general, I think the project would benefit from a normative system verilog definition of the interface, and that would have to have ifdefs. I could imagine maybe having a script that will generate an ifdef free version for specific parameterizations.

why do you want to avoid ifdefs?

Silabs-ArjanB, 2023-05-15:
We definitely want to avoid custom preprocessing and additional template languages. Ifdefs often lead to problems in various tools.

zarubaf, 2023-05-15:
You could use `type` parameters to avoid ifdefs.

```
core #(
  parameter type x_if_req_t = logic,
  parameter type x_if_rsp_t = logic
) (
  output x_if_req_t x_if_req_o,
  input x_if_rsp_t x_if_rsp_i
);
```

christian-herber-nxp, 2023-05-15:
@Silabs-ArjanB: Besides the ifdef question, there is also a more important one: Is it the intention of the specification, that e.g. a core that does not have dualread capability has the interface. The current specification would suggest yes, but i would assume the intention is no.

Silabs-ArjanB, 2023-05-15:
The specification says 'yes' and the intention was 'yes' as well.

zarubaf, 2023-05-15:
> @zarubaf thanks for the suggestion. I don't see a way to make a signal disappear though with typed parameters. The `dualread` signal would really have to disappear from the interface for `X_DUALREAD = 0`

By using a struct that does not contain the signal you can avoid having it in the port list. But at the end of the day, it is all hacks because the language is so severely limited.

christian-herber-nxp, 2023-05-15:
> The specification says 'yes' and the intention was 'yes' as well.

ok, thanks for the clarification

christian-herber-nxp, 2023-05-15:
> By using a struct that does not contain the signal you can avoid having it in the port list. But at the end of the day, it is all hacks because the language is so severely limited.

that sounds way too hacky. if you start having multiple of such options, you need to offer a high number of permutations of the same struct.

christian-herber-nxp, 2023-05-17:
@Silabs-ArjanB after giving this some thought, I think you should not shy away from using ifdef statements in the verilog examples. After all, this is a specification and whatever serves the specification and its readers best should be the way to go. Giving a potential user an indication about how this could be tailored is valuable. 
I acknowledge your point about tools, but this is a specification and not a synthesizable IP.

Another example of features that should probably be optional is speculative load/stores. CV32E40X does allow a coprocessor to request a speculative load/store, but completely ignores the mem_req.spec signal. A co-processor, which relies on this getting special treatment will not be compatible, but it is still possible to integrate it.

Silabs-ArjanB, 2023-05-18:
>>  I think you should not shy away from using ifdef statements in the verilog examples. 

I agree that this is a spec as opposed to a synthesizable IP, but for us ifdefs are simply a no-go, so we definitely do not want to advertise Verilog examples using that.

>> Another example of features that should probably be optional is speculative load/stores. CV32E40X does allow a coprocessor to request a speculative load/store, but completely ignores the mem_req.spec signal. 

We did not yet try to implement the complete XIF spec yet in the CV32E40X, so ignoring the spec signal does not imply anything. Surely we did not intend on treating the spec signal as optional and as far as I am concerned we should leave as little such optionality in the XIF spec as possible to avoid a bunch of coprocessors and processors that cannot operate together. I don't think it is a big burden on a processor to support that signal.

christian-herber-nxp, 2023-05-22:
@Silabs-ArjanB : agree that flexibility should be limited to have a high number of co-processors and cores interoperable. however, this may limit the adoption with CPUs where this configuration is not suitable. there are levels of flexibility which can also be implemented in a co-processor. this will enhance the overall support rather than fragment it. 
If there are implementations in the open that only implement parts of the specification, i would prefer if the interface specification contains sufficient parameters to make sure a CPU or coprocessor can declare its capabilities and expectations in a clear manner.

christian-herber-nxp, 2024-02-21:
closing this issue. We have some level of parameterization in place. No intention to go beyond this to limit the possible variants of CV-X-IF

closed 2024-02-21 by christian-herber-nxp