openhwfoundation/core-v-xif #69: Some parameters need clarification

christian-herber-nxp, 2023-05-17:
In particular, `X_MISA` and `X_ECS_XS` lack sufficient explanation to understand what they would be used for in a coprocessor. They also do not quite appear in the rest of the specification like other parameters.

christian-herber-nxp, 2024-02-23:
@Silabs-ArjanB : The spec currently describes the ``X_ECS_XS`` parameter as:

> Initial value for ``mstatus.XS``.

I assume that is actually a parameter of the CPU, less the interface, as the coprocessor really should care about this?
I am not sure what text could be added to describe what this parameter is about.

christian-herber-nxp, 2024-02-23:
So what is the requirement towards the interface? Is there e.g. a requirement for the extensions to be in a specific state? I see an issue that a copro will not be able to update XS on its own initiative, e.g. after completing its reset sequence. It needs an instruction offloaded in order to write to XS.
For me, this parameter raises more questions at this point than it answers.

jquevremont, 2024-02-26:
When reading the ``X_MISA`` description, I am a bit confused.

The specification states:

> MISA extensions implemented on the eXtension interface.

So I could understand that the value is defined based on the extension(s) provided by the coprocessor.

But the document also states:

> The CPU determines the legal values for this parameter.

So I could understand that the value is defined based on the extension(s) supported natively supported by the processor.

Could you clarify? An example could help (e.g. an RV32IMAC core extended a coprocessor supporting the F extension). A second example could illustrate a custom extension.

My expectation is that ``X_MISA`` encodes the extension(s) provided by the coprocessor and that the processor will logically OR it with its own ``MISA``.

christian-herber-nxp, 2024-02-27:
> So what is the requirement towards the interface? Is there e.g. a requirement for the extensions to be in a specific state? I see an issue that a copro will not be able to update XS on its own [...]

@Silabs-ArjanB : Could you have a look at this comment? We need to come up with some descriptive text of what a Coprocessor/CPU is supposed to do with the ECS parameter

Silabs-ArjanB, 2024-02-27:
X_ECS_XS is a parameter to the processor only. It defines the initial (i.e. reset) value for mstatus.xs. This value is something that can only be known by the person integrating a processor with a given coprocessor and therefore it needs to be a parameter to the processor.

Silabs-ArjanB, 2024-02-27:
These things are different indeed. In the end we need to be able to make sure that the mstatus and misa CSRs reflect the combination of a processor and coprocessor, that's why we need these parameters.

christian-herber-nxp, 2024-02-27:
I will merge the PR and thus close this issue, as we can tackle ECS through the more recent issue.

closed 2024-02-27 by christian-herber-nxp