openhwfoundation/core-v-xif #73: List dependencies for building documentation in readme

christian-herber-nxp, 2023-07-04:
Dependencies should be explicit, not for a user to find out in trial and error. From what I can see, those packages are needed
```
pip install recommonmark
pip install sphinxcontrib-svg2pdfconverter
pip install sphinx_github_changelog
```

christian-herber-nxp, 2023-07-04:
Hijacking this one, before opening another issue, as this is likely dependency related. I ran into the following problem:

```
Running Sphinx v6.2.1
WARNING: logo file '../images/openhw-landscape.svg' does not exist
building [mo]: targets for 0 po files that are out of date
writing output...
building [html]: targets for 4 source files that are out of date
updating environment: [new config] 4 added, 0 changed, 0 removed
reading sources... [ 75%] preface
Exception occurred:
  File "C:\Users\nxa23603\AppData\Local\Programs\Python\Python310\lib\site-packages\sphinx_github_changelog\changelog.py", line 44, in compute_changelog
    return no_token(changelog_url=options["changelog-url"])
KeyError: 'changelog-url'
The full traceback has been saved in C:\Users\nxa23603\AppData\Local\Temp\1\sphinx-err-jnwmcfi8.log, if you want to report the issue to the developers.
```

So i went about and updated Sphinx, which leads to this issue:

```
Installing collected packages: sphinx
  Attempting uninstall: sphinx
    Found existing installation: Sphinx 6.2.1
    Uninstalling Sphinx-6.2.1:
      Successfully uninstalled Sphinx-6.2.1
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
sphinx-rtd-theme 1.2.2 requires sphinx<7,>=1.6, but you have sphinx 7.0.1 which is incompatible.
Successfully installed sphinx-7.0.1
```

In conclusion, it seems you would need to specify exact versions for all the packages (and sphinx itself). Of course, making things robust with respect to package version would be even better

christian-herber-nxp, 2023-07-04:
finally, my recommendation would be to add a requirements.txt file (see https://note.nkmk.me/en/python-pip-install-requirements/) that specifies the valid ranges of the respective packages/extensions

MikeOpenHWGroup, 2023-07-04:
As recommended by RTD, the requirements file is located at [core-v-xif/docs/requirements](https://github.com/openhwgroup/core-v-xif/blob/cd906260418108343a1115cf443a7f86af3c7d04/docs/requirements.txt#L1).

Note that I am not the "code-owner", nor a significant contributor to this repo, so I will defer the final resolution to others.

MikeOpenHWGroup, 2023-07-04:
Hi @christian-herber-nxp, if you submit a PR to resolve this issues I can test it.

christian-herber-nxp, 2024-02-21:
a readme was added to point to requirements.txt with instructions about how to build the documentation

closed 2024-02-21 by christian-herber-nxp