openhwfoundation/core-v-xif #76: rs_valid needs clarification

christian-herber-nxp, 2023-09-13:
rs_valid is currently a bit underspecified. 
- the use of those signals is not very explicit. The spec does state:
  > A coprocessor can (only) accept an offloaded instruction when: The required source registers are marked valid by the offloading core (issue_valid is 1 and required bit(s) rs_valid are 1).
  
  but it is not quite clear what required means.
- the definition of rs_valid is unclear for X_RFR_WIDTH=64, XLEN=32. What does it mean, if a bit is asserted? Without further detailing, I would assume this means that the entire register pair is valid. This would be a huge performance hit for instructions that do not make use of pairs, as they would still have to wait for the entire pair to be ready. My suggestion would be to change the Type from `logic [X_NUM_RS-1:0]` to `logic [X_NUM_RS+X_DUALREAD-1:0]` to be able to specify the validity of each individual x register

christian-herber-nxp, 2023-09-13:
@Silabs-ArjanB @davideschiavone what's your opinion?

davideschiavone, 2023-09-14:
I agree with both statements.

We need to clarify that if the coprocessor doesn't need the register, it can ignores the valid bit.

And I agree with your proposal regarding the dual read

Silabs-ArjanB, 2023-09-14:
Hi,

>> I would assume this means that the entire register pair is valid.

Yes that is what it means (but I agree it is not that clear from the spec).

Also agreed that this is not efficient for instructions not using dual read and that we therefore need a valid bit per individual operand that can potentially be accessed.

>> change the Type from logic [X_NUM_RS-1:0] to logic [X_NUM_RS+X_DUALREAD-1:0] 

Agreed with the _idea_ but above implementation does _not_ give what you want (given the current encoding of X_DUALREAD). Maybe we should just change the definition of X_NUM_RS to include dual read alternatives as well.

christian-herber-nxp, 2023-09-14:
> Agreed with the idea but above implementation does not give what you want (given the current encoding of X_DUALREAD).

I don't see why.

> Is dual read supported? 0: No, 1: Yes, for rs1, 2: Yes, for rs1 - rs2, 3: Yes, for rs1 - rs3.

This means, for each pair that is supported, one additional bit will be added to the vector.

> Maybe we should just change the definition of X_NUM_RS to include dual read alternatives as well.

`X_NUM_RS` is defined as the number of read ports. I think this is precise as it is right now. The name is a bit confusing, I assume something changed here. Renaming to `X_NUM_RP` might make sense.
You could add a parameter `X_NUM_RXREGS` for the number of readable x-registers, but would just be a shorthand for `X_NUM_RS+X_DUALREAD`

Silabs-ArjanB, 2023-09-14:
>> This means, for each pair that is supported, one additional bit will be added to the vector.

You are right; that works. I erroneously looked at the encoding of the dualread signal in x_issue_resp_t.

PR #78 (2023-09-19): Fixed performance issue with register pairs
When using register pairs, the validity of source operands was signalled per register pair only. Thus, instructions that only use a single register for an operand have to wait for the entire pair to be valid and might be stalled unnecessarily.

Fixes #76
commit d31b5e72 (2023-09-19): Fixed performance issue with register pairs

When using register pairs, the validity of source operands was signalled
per register pair only. Thus, instructions that only use a single
register for an operand have to wait for the entire pair to be valid and
might be stalled unnecessarily.

Fixes #76
closed 2023-09-19 by davideschiavone