openhwfoundation/core-v-xif #84: Clarify the mode should follow the effective privilege level

christian-herber-nxp, 2023-10-18:
Currently, mode is specified to be the privilege level of the CPU. However, it should really be the effective privilege level. If machine mode software changes its effective privilege mode by setting mstatus.MPRV, instructions should be offloaded at the effective level.

jklockars, 2023-10-18:
It is not as simple as that. While MPRV is the "effective privilege level", that is only relevant for load/store, which are also affected by SUM and MXR (more?). But other instructions may be required to cause synchronous faults depending on the actual CPU mode (including virtualization), and there are also things like filtered HPM counters which could be affected.
We may need to pass a whole bunch of bits as "mode".

christian-herber-nxp, 2023-10-19:
Actually, I am not so sure it is possible to already know the state of MPRV/SUM/MXR at the time of issuing an instruction to the copro in general. This is a tricky one.

jklockars, 2023-10-19:
True. Depending on how "issue" is defined for core-v-xif.
I suppose one option would be to require the "mode state" to be known before the actual issueing. Otherwise it would (AFAICS) either need to be handled only in the main pipeline, or somehow passed in later. In either case, co-pro faults depending on "mode state" could be challenging.

In our NOEL-V, the main pipeline is guaranteed to never contain instructions in different privilege modes, and there are some interlocks between various CSR writes and certain types of instructions. Some of this would need special handling to be able to deal with core-v-xif instructions that might depend on CSR and/or pipeline state. I am sure there are micro-architectures where things are far more complicated.

For our FPU, we get away with passing in only (well, also rounding mode) the privilege mode (needed for filtering), since load/store (which would have required MPRV etc) is handled by our main pipeline.
(The data transfers for FPU load/store are handled in exactly the same way as transfers between the FPU and integer register files.)
Similar handling could probably be done for some core-v-xif units.

Silabs-ArjanB, 2023-10-20:
The mode transferred needs to be the privilege level of the CPU (not related to MPRV). The MPRV setting is taken care of by passing loads/stores through the CPU's LSU.

christian-herber-nxp, 2023-10-20:
> The mode transferred needs to be the privilege level of the CPU (not related to MPRV). The MPRV setting is taken care of by passing loads/stores through the CPU's LSU.

Fair enough. In that case, is mem_req.mode really needed? The LSU should be able to deduct the right privilege level.

Silabs-ArjanB, 2023-10-20:
If the coproc would not pass that mode then the processor would need to keep track of the privilege mode for offloaded loads/stores, which is just complicating things.

christian-herber-nxp, 2023-10-20:
right, that's the point. I don't understand why keeping track of e.g. mstatus.MPRV is easier than keeping track of the mode

Silabs-ArjanB, 2023-10-20:
Okay, if you would have multiple offloaded loads/stores simultaneously, then it would indeed be easier to send both CPU privilege and MPRV over the issue interface to the coprocessor, and then receive back the effective privilege of a load/store over the memory interface from the coprocessor.

JeanRochCoulon, 2023-10-20:
The discussion is interesting to evaluate what should be transmitted between CPU and copro. But at the end, why not following the AMBA bus specification, that is to say define a field called USER to export CSR information. This processor change is not intrusive in term of bug insertion because we do not alter the proc functionality. We can consider the solution remains generic, without major proc customization. And on the other end, copro can access to the CSR information.

christian-herber-nxp, 2023-10-23:
As a side question, I would like to check on the intended scope of cv-x-if: Is the intention to support privileged extensions? Intuitively, I would say no. In that case, the mode information is only needed for load stores and nothing else. And then my suggestion would be to convert it to the effective privilege, rather than the actual one.

Silabs-ArjanB, 2023-10-24:
I would say 'yes'. A coprocessor needs the mode information for example to decide if a CSR access is legal or not. Also some regular instructions might only be legal in a specific privilege mode.

christian-herber-nxp, 2023-10-24:
> I would say 'yes'. A coprocessor needs the mode information for example to decide if a CSR access is legal or not. Also some regular instructions might only be legal in a specific privilege mode.

then by definition you assume that some extensions are privileged. For unprivileged extensions, mode must not be considered to access CSRs or determine legality of instructions.

jklockars, 2023-10-24:
I never considered extensions having their own CSRs, but I suppose that is unavoidable.

Do we intend to make CV-X-IF general enough that _any_ RISC-V extension that does not affect other things (such as F/D/Q/V/B/etc, but not for example H/Sscofpmf/Zicfilp) should be implementable that way?

That would mean that any accesses to CSR addresses unknown to the general pipeline must instead be passed to the coprocessor. Do we then add some kind of interlock-interface to be able to guarantee that things behave properly, or perhaps instead enforce pipeline hold or flush for each such CSR access?

What about extensions that need access to some bits from standard CSRs (xSTATUS, xENVCFG etc)?

christian-herber-nxp, 2023-10-24:
@jklockars: The expectation is that a CSR instruction that goes to an unimplemented CPU traps. Instructions that trap should be attempted for offloading. There is currently no pipeline flush or anything foreseen. Custom CSRs in the copro should not affect the main CPU. There is precedent for bits needed from xstatus (see ecs in the specification). It's not a great solution, but nobody has a better one.
However, I would like to stress that I do not see the need for privileged CSRs to be implemented in a copro.

christian-herber-nxp, 2023-10-25:
@Gchauvon, @zarubaf  : Does it make sense to offload SUM and MXR bits of mstatus in issue interface for CVA6? Or will the LSU be able to have the right setting for each load/store

closed 2023-11-13 by christian-herber-nxp