openhwfoundation/core-v-xif #144: Ambiguous ID requirement description

flegert, 2024-02-07:
The definition of the allowed ID values contains the following snippet:

> The id values are required to be incremental wrapping for sequential instructions, but do not necessarily need to be continuous.

The sentence leaves room for interpretation. "Sequential instructions" should be clearly defined (Sequential in a sense that no native CPU instructions are in-between? Sequ. from CPU perspective? ...)
Due to the way the sentence is written it could also be misinterpreted which IDs can be "continuous" and which don't have to be. 

Maybe it would also help to clearly define what is meant with "incremental wrapping" and how potential overflows of IDs have to be handled.

flegert, 2024-02-09:
Thank you for the explanations. One potential improvement I would suggest might be something like this:
> The id values are required to be incremental wrapping for sequential instructions, but do not necessarily need to be continuous **otherwise** / **between sequential offloads**.

This would prevent possible misunderstandings (without it one might interpret the id values to be "incremental wrapping" but "not continuous" at the same time)

For the term "sequential instruction", a note might help. I'm not sure how I would formulate that either, but maybe something along those lines but more formal:
> Note: Sequential instructions relate to offloaded instructions that appear in order without any intermediate instructions recognized by the CPU.

christian-herber-nxp, 2024-02-09:
Now I am not sure if we are on the page. By non-continuous, it is meant that the increment does not have to be 1.

flegert, 2024-02-09:
As I understood the definitions, _non-continuous_ means increments by 1, 2, 3, ... are possible, while _incremental wrapping_ means only increment by 1, and go from 0x...FF -> 0x...00.

As I understood the sentence:
"For sequential instructions, the id value must be incremental wrapping (-> and therefore must also be continuous). If the instructions are not sequential (e.g., if inbetween there is an instruction handled solely by the CPU), the id value can be non-continuous."

I hope I did not interpret that wrong.

If it is correct, than in my opinion the sentence still might be interpreted in other, _self-contradicting_ ways, e.g.:
"The id values for sequential instructions must not be continuous but must be incremental wrapping." 

To put it short, the sentence seems correct but may be misleading.
That's why I suggested adding "otherwise" or "between sequential offloads" at the end and separatly define "sequential instruction" to leave less room for such wrong interpretations.

Feel free to object though, maybe I'm overemphasizing this, also I don't want to waste your time on topics that may be too minor right now.

closed 2024-02-21 by christian-herber-nxp