openhwfoundation/core-v-xif #145: Clarification of valid-ready handshake dependency

flegert, 2024-02-07:
The manual states:
 
> 4.9 Handshake dependencies
> ...
> The valid signal of a transaction shall not be dependent on the corresponding ready signal.
> ...
> Note: The use of the words depend and dependent relate to logical relationships, ...
 
However, as I understand it, in some way each valid signal has to be logically dependent on the ready signal. For example, a coprocessor generating a result transaction would set result_valid = 1, keep it high until CPU sets result_ready = 1, and then resets result_valid = 0 in the following cycle i no further instruction is processed at the moment.

Maybe it would help to define what "logically dependent" exactly means in this context?

christian-herber-nxp, 2024-02-08:
@Silabs-ArjanB I think your input would be valuable here.
I agree that there should actually be a logical relationship, in a sense that the deassertion of ``valid`` may be a consequence of the assertion of ``ready``. I would assume that by logical relationships both combinatorial and sequential logic relationships are meant

Silabs-ArjanB, 2024-02-08:
The requirement is for 1 transaction. The supposed counter example given above is about 2 transactions.

flegert, 2024-02-09:
You are right, I missed the transaction constraint! Apologies for the oversight, given this I would suggest closing the issue.

closed 2024-02-09 by christian-herber-nxp