openhwfoundation/core-v-xif #150: Specification review (several remarks)

jquevremont, 2024-02-16:
I have reviewed the PDF of 1.0.0-rc.1. No blocking remark. Most are minor to ease reading.

General:
- The copyright mention appears twice in the document.
- A glossary would be nice (e.g. for acronyms like RTL, ALU...)

Changelog:
- The 0.1.0 and 0.2.0 version in the Changelog do not align with the 0.1.0 and 0.2.0 GitHub releases.
- As there have been early implementations of CV-X-IF (CVA6, CV32E40X, Vicuna...), could you add a note about the compatibility of CV-X-IF v1.0.0 with former GitHub releases?

Chapter 3:
- "Bit Manipulation" is now Zb*, no more B.
- 3.1 I suggest to also mention that CV-X-IF is an optional interface of CVA6 (as you refer to CVA6 in section 5.4).

Chapter 4:
- 4.2 As of now, most RISC-V and CORE-V cores only have one hart. In such case, ``hartid`` looks superfluous.
- 4.5 Typical waveforms for the various interfaces would help the understanding of the interfaces.
- 4.5.1 If the ``id`` increment is greater than 1 and ``id`` wraps, the coprocessor cannot determine the order of instructions with ``id``. Then why is ``id`` required to increment?
- 4.5.3 I suggest to state that the compressed instructions expands instructions without executing them.
- 4.5.3 I suggest to mention that compressed instructions that translate to several non-compressed instructions (like Zcmp) or do not have equivalent non-compressed instructions (like Zcmt) are not supported.
- 4.5.4, 4.5.6, 4.7 Everyone knows what it is, but ``clk`` is been defined.

Chapter 5:
- 5.2 You introduce the possibility of several coprocessors attached to a CPU, but do not explain how this can be realized (attached to the same CV-X-IF? several CV-X-IF sockets?)
- 5.3 May I suggest to copy the figure CV32E40X timing budget  to the document, as the URL could be broken in the future?

christian-herber-nxp, 2024-02-19:
> I have reviewed the PDF of 1.0.0-rc.1. No blocking remark. Most are minor to ease reading.
> 
> General:
> 
> * The copyright mention appears twice in the document.
it was requested for it to appear on every page. The current way to have it is the closest I can do it.
> * A glossary would be nice (e.g. for acronyms like RTL, ALU...)
ok, agree
> 
> Changelog:
> 
> * The 0.1.0 and 0.2.0 version in the Changelog do not align with the 0.1.0 and 0.2.0 GitHub releases.
What is the decrepency?
> * As there have been early implementations of CV-X-IF (CVA6, CV32E40X, Vicuna...), could you add a note about the compatibility of CV-X-IF v1.0.0 with former GitHub releases?
> 
> Chapter 3:
> 
> * "Bit Manipulation" is now Zb*, no more B.
There is currently a fast-track extension for B ongoing. While not yet ratified, I would propose to let this one slide, as it will be correct in the future.
> * 3.1 I suggest to also mention that CV-X-IF is an optional interface of CVA6 (as you refer to CVA6 in section 5.4).
ok.
> 
> Chapter 4:
> 
> * 4.2 As of now, most RISC-V and CORE-V cores only have one hart. In such case, `hartid` looks superfluous.
Almost anything can be superfluous based on the configuration. In that case, it will have a constant value and will be optimized away. Please also note that a multi-processor system sharing a coprocessor also has multiple harts.

> * 4.5 Typical waveforms for the various interfaces would help the understanding of the interfaces.
Agree. It is one the wishlist, but so far nobody has contributed this.
> * 4.5.1 If the `id` increment is greater than 1 and `id` wraps, the coprocessor cannot determine the order of instructions with `id`. Then why is `id` required to increment?
The coprocessor will be able to track that the ids wrapped
> * 4.5.3 I suggest to state that the compressed instructions expands instructions without executing them.
ok. should be added.
> * 4.5.3 I suggest to mention that compressed instructions that translate to several non-compressed instructions (like Zcmp) or do not have equivalent non-compressed instructions (like Zcmt) are not supported.
it is supported to have instructions that do not have an equivalent 32-bit instruction. It is possible to decompress into a 32-bit instruction that is only known to the coprocessor.
> * 4.5.4, 4.5.6, 4.7 Everyone knows what it is, but `clk` is been defined.
ok
> 
> Chapter 5:
> 
> * 5.2 You introduce the possibility of several coprocessors attached to a CPU, but do not explain how this can be realized (attached to the same CV-X-IF? several CV-X-IF sockets?)
this is an intentional decision. one would have to create another specification to define how this should be realized.
> * 5.3 May I suggest to copy the figure CV32E40X timing budget  to the document, as the URL could be broken in the future?
Ok. I would prefer to not copy this in, but rather to make the link more robust (i.e. refer to a tagged version of the document so the link is stable)

christian-herber-nxp, 2024-02-19:
@jquevremont please check if my answers are ok. In general, while a bit more work, the best way to go about this is to have one issue per problem, then each item can be tracked separately. There will also be a separate pull request per item, thats where it will help a lot.

jquevremont, 2024-02-20:
@christian-herber-nxp, may I suggest that you mention this preference for 1 issue per problem with a temporary mention in the README file?

Regarding your answers:

> > Changelog:
> > 
> > * The 0.1.0 and 0.2.0 version in the Changelog do not align with the 0.1.0 and 0.2.0 GitHub releases.
> 
> What is the decrepency?

In the Changelog in the PDF version, 0.1.0 is dated 2024-02-13 and 0.2.0 is dated 2024-02-13.

In https://github.com/openhwgroup/core-v-xif/tags, there are tags 0.1.0 dated 2021-08-30 and core-v-xif_v0.2.0 dated 2022-04-22.

> > * As there have been early implementations of CV-X-IF (CVA6, CV32E40X, Vicuna...), could you add a note about the compatibility of CV-X-IF v1.0.0 with former GitHub releases?

In this remark, I meant the "old" GitHub releases/tags in https://github.com/openhwgroup/core-v-xif/tags, not the recent releases you managed.

> > * 4.5.1 If the `id` increment is greater than 1 and `id` wraps, the coprocessor cannot determine the order of instructions with `id`. Then why is `id` required to increment?
> 
> The coprocessor will be able to track that the ids wrapped

I suggest to make it explicit that the increment shall be stritly lower than ``2**X_ID_WIDTH``.

christian-herber-nxp, 2024-02-20:
> @christian-herber-nxp, may I suggest that you mention this preference for 1 issue per problem with a temporary mention in the README file?
>
> Regarding your answers:
>
>
> In the Changelog in the PDF [...]

I see. This is true, as I had to fix those releases after creating the release candidate pdf as their RTD build was broken. Not nice, but any new version of the specification will have the changelog synchronized.

> In this remark, I meant the "old" GitHub releases/tags in https://github.com/openhwgroup/core-v-xif/tags, not the recent releases you managed.
>
>
> I suggest to make it explicit that the increment [...]

In fact, I don't think that would be an issue. If not all instructions are presented to a coprocessor, that is actually a realistic and totally fine scenario. Also, the width of the ID is required to be sufficiently large, so that when you would actually have such an increment, the old instruction is guaranteed to have retired.

jquevremont, 2024-02-21:
After our meeting, I have replaced these issues with issues #153 to #160, with one problem per filed issue. Therefore, I close this issue.

closed 2024-02-21 by jquevremont