openhwfoundation/core-v-xif #156: Overlapping 0.1.0 and 0.2.0 tags

jquevremont, 2024-02-21:
In the Changelog in the PDF version, 0.1.0 is dated 2024-02-13 and 0.2.0 is dated 2024-02-13.

In https://github.com/openhwgroup/core-v-xif/tags, there are tags 0.1.0 dated 2021-08-30 and core-v-xif_v0.2.0 dated 2022-04-22.

christian-herber-nxp, 2024-02-23:
There are multiple things at play here. The changelog does not refer to the date of the commit that the tag points to, but the day the release was created.
The core-v-xif_v0.X.0 tags are unchanged from their original state, and should remain so, so that any external references to those are intact. In an effort to improve the changelog, I updated the releases. Just changing the text will update the date (I learned that then). I also introduced new tags v0.2.0 and v0.1.0, to have a cleaner labeling. 
Meanwhile, I even had to update v0.2.0 with one commit to fix the build. 

Long story short: Old tags are pointing to where they always pointed. New tags and cleaner changelog is in place now, and builds are working. Consequence is that we have updated releases, with new dates. It's not perfect, but now that we have fixed all builds etc., we should have a solid base going forward to avoid any such disturbances in the future.

So, nothing to fix from my side.

jquevremont, 2024-02-23:
Thanks for clarifying this difference of dates between tags and releases. I have learnt something about GitHub. Indeed, nothing to fix and we'll need keep this unobvious datestamps.

closed 2024-02-23 by jquevremont