openhwfoundation/core-v-xif #157: Document compatibility of 1.0.0 specification with core-v-xif_v0.2.0 release

jquevremont, 2024-02-21:
As there have been early implementations of CV-X-IF (CVA6, CV32E40X, Vicuna...), could you add a note about the compatibility of CV-X-IF v1.0.0 specification with [core-v-xif_v0.2.0](https://github.com/openhwgroup/core-v-xif/releases/tag/core-v-xif_v0.2.0) GitHub release dates 2022-04-22?

christian-herber-nxp, 2024-02-23:
I tried to list the relevant changes in the changelog of the v1.0.0-rc.1 release. Tag core-v-xif_v0.2.0 is equivalent to v0.2.0 with the only difference that v0.2.0 has a fixed build.

Does this clarify? Any action needed?

jquevremont, 2024-02-23:
Thanks for pointing this. I can read it in https://github.com/openhwgroup/core-v-xif/releases/tag/v1.0.0-rc.1. I was looking for this information in the PDF file.

christian-herber-nxp, 2024-02-23:
I see. It is also here in the html: https://docs.openhwgroup.org/projects/openhw-group-core-v-xif/en/latest/preface.html#changelog
No idea why sphinx does not pick that up for the pdf version. Seems to be a limitation in Sphinx. It is the same issue in CV32E40P documentation. We could file in an issue with Sphinx, to have an option to include the text, as that would be really useful.

christian-herber-nxp, 2024-02-23:
Created this ticket https://github.com/ewjoachim/sphinx-github-changelog/issues/122

jquevremont, 2024-02-23:
Alternatively, this could be written in section 3.1 History.

christian-herber-nxp, 2024-02-23:
> Alternatively, this could be written in section 3.1 History.

We can write a manual changelog, that wouldn't be an issue. But the point is to have no duplication of anything, and this integration with github allows this, so I think we should stick to it but make it work.

jquevremont, 2024-02-23:
I understand. Until the Sphinx ticket , I suggest to add a manual changelog as a temporary workaround. And a [core-v-xif](https://github.com/openhwgroup/core-v-xif) issue could be created in order to remove the workaround when the issue is fixed in Sphinx.