openhwfoundation/core-v-xif #177: Interface requirements to maximize interoperability

jquevremont, 2024-02-26:
For the sake of maximum compatibility between cores and coprocessors and clarity for the designers.

What are the mandatory and optional interfaces for a coprocessor? E.g. if a coprocessor does not support compressed instructions, is it required to implement the compressed interfaces?

Same question on the core side. E.g. if a coprocessor does not issue instructions speculatively, is it required to implement the compressed interfaces?

I have not obviously found the answer in the specification. I see the answer fit for the Appendix.

christian-herber-nxp, 2024-02-26:
Currently, only the memory interface is optional.
I am in favor of clearly parameterizing the scenarios you mentioned. I did face some opposition for adding additional parameterization to the interface. So I do not really have a solution for your issue.

jquevremont, 2024-02-27:
If the issue is controversial, let's close it.

closed 2024-02-27 by jquevremont