openhwfoundation/core-v-xif #182: [Review Comment]: X_ISSUE_REGISTER_SPLIT definition

ntribie-bosch, 2024-03-01:
### Comment

As far as I understand, X_ISSUE_REGISTER_SPLIT is a parameter defined by the CPU implementation, and passed to the coprocessor.

When  X_ISSUE_REGISTER_SPLIT=0 :  “_In this case, the CPU will speculatively provide all possible source registers_” :  I guess that if the provided registers value were wrong , the CPU will send a kill request to the coprocessor using the x_commit_t interface ? 

When  X_ISSUE_REGISTER_SPLIT=1 .  How does the CPU supporting this mode can know that “ _the coprocessor is able to accept multiple issue transactions before receiving the registers_ “ ?  It would be nice if the coprocessor could have a signal / parameter to restrict this behavior to the ISSUE_REGISTER_SPLIT=0 mode , for back compatibility reasons. That is, getting the instruction and its operands values at the same time ,  at the expense of performance.

### Proposed Resolution

N/A

### Addition Info

N/A

christian-herber-nxp, 2024-03-01:
> _ISSUE_REGISTER_SPLIT=0 : “_In this case, the CPU will speculatively provide all possible source registers_” : I guess that if the provided registers value were wrong , the CPU will send a kill request to the coprocessor using the x_commit_t interface ?
> 

The CPU only provides correct values when the according valid signal is asserted. Thus, no need to kill.

> When X_ISSUE_REGISTER_SPLIT=1 . How does the CPU supporting this mode can know that “ _the coprocessor is able to accept multiple issue transactions before receiving the registers_ “ ? It would be [...]

If the coprocessor is not able to buffer issue offloads, it will not accept another issue before receiving the registers. No accepting would be achieved by not asserting the issue_ready signal. This would stall the CPU until the registers have been provided for the preceding instruction.

ntribie-bosch, 2024-03-01:
Ok, thanks for your quick feedback . So we can solve back – compatibility.

Sincères salutations / Best regards,

Nicolas Tribie

Mobility Electronics, IC Architecture (ME-IC/PAY1.2)
Robert Bosch (France) SAS | 1180 route des Dolines | 06560 Valbonne-Sophia Antipolis | FRANCE | www.bosch.fr<http://www.bosch.fr/>
***@***.******@***.***>

Registered Office: 32 avenue Michelet - 93404 St Ouen cedex France
Société par Actions Simplifiée au capital de 140.400.000 € - SIREN 572 067 684 RCS Bobigny
N° TVA France : FR 09 572 067 684
​

closed 2024-03-01 by christian-herber-nxp