openhwfoundation/core-v-xif #183: [BUG] v1.0.0-rc.1 : exc, excode, dbg and err still there

pascalgouedo, 2024-03-04:
### Is there an existing CV-X-IF Bug for this?

- [X] I have searched the existing bug issues

### Task Description

Even if this issue #151 is marked as merged and closed with PR #152 those 4 signals are still in v1.0.0-rc.1.

I completely understand that dbg and err should be moved to Memory interface.

But I don't get the point of moving exc/exccode as well.
I think any non-memory instruction should be able to report an exception related to its own architectural definition if wanted.

### Proposed resolution

Remove dbg and err.

Explain how to signal exceptions to the CPU if exc* are moved to Memory interface.

### Description of Done

-

closed 2024-03-04 by christian-herber-nxp

pascalgouedo, 2024-03-04:
What about exc/exccode? Still moved to memory interface?
If local XS is off, how to answer with illegal instruction error without memory interface?

christian-herber-nxp, 2024-03-04:
By not accepting the instruction. Then nobody will accept it an an illegal instruction trap will happen. But maybe that's a fair point to put into the spec explicitly