openhwfoundation/core-v-xif #184: [Review Comment]: Interface dependencies

pascalgouedo, 2024-03-04:
### Comment

Why a register interface transaction is required for each issue interface transaction?

If a coprocessor instruction is only using its own custom CSRs and its own custom register file, this register interface transaction is useless. It can accept an issue with all other signals set to 0 meaning no registers read/write on CPU side.

### Proposed Resolution

Replace
"Every issue interface transaction has an associated register interface transaction."
by
"Every issue interface transaction has an associated register interface transaction only if any issue response read/write signal is set to 1."

pascalgouedo, 2024-03-04:
In fact this is linked with how coprocessor sees mstatus.xs value. Because it should drop any instruction executed in a new user extensions requiring state management if XS is off and report an illegal instruction exception.

christian-herber-nxp, 2024-03-04:
Same argument could be made for a result transaction. If you are not writing back, why even have a transaction? I think the logic taken here is to always have those transactions to simplify state machines.

pascalgouedo, 2024-03-04:
Like OBI or any other protocol, result/valid response is necessary even for write to have the confirmation the whole transaction is finished on the final target.
Intermediate interface transactions don't have the same outcome and they generate unnecessary traffic and then unnecessary power consumption.
But I agree it simplifies state machines.

christian-herber-nxp, 2024-03-06:
Decision from this weeks meeting is to keep having 1 register transaction per issue transaction

closed 2024-03-06 by christian-herber-nxp