openhwfoundation/core-v-xif #200: [Review Comment]: Retraction of compressed itf transactions not flexible enough

Silabs-ArjanB, 2024-03-15:
### Comment

[image]

### Proposed Resolution

See above

christian-herber-nxp, 2024-03-15:
@Silabs-ArjanB The reason why we removed it was, that the assumption was that you would just change instr without de-asserting valid. Practically, that will surely work. That would be inline with the last sentence of your comment.

Let me know the path you prefer.

Silabs-ArjanB, 2024-03-15:
I am fine with just changing instr while keeping valid asserted (without introducing an id), but this is not allowed as per above rules as it would require the processor to change hartid.