openhwfoundation/core-v-xif #201: [Review Comment]: Compressed itf only for compressed instructions

Silabs-ArjanB, 2024-03-15:
### Comment

[image]

### Proposed Resolution

See above

christian-herber-nxp, 2024-03-15:
did we not discuss this in one of the meetings and concluded it was not necessary to put such a restriction into the spec?

Silabs-ArjanB, 2024-03-15:
I forgot the conclusion and I don't think it matters much, but we need to make either choice explicit (I don't care what choice)

christian-herber-nxp, 2024-03-15:
as the coprocessor has to anyway inspect the LSBs, I would put the requirement on the copro. At the same time, the CPU will likely anyway be aware of the instruction length, as it has to obtain 16B from the prefetch buffer. But putting the restriction on the copro is I think the smaller restriction

closed 2024-03-19 by christian-herber-nxp