openhwfoundation/core-v-xif #204: [Review Comment]: Register interface underspecified

Silabs-ArjanB, 2024-03-15:
### Comment

[image]

[image]

[image]

[image]

[image]

### Proposed Resolution

No proposed resolution

christian-herber-nxp, 2024-03-18:
> ![image](https://private-user-images.githubusercontent.com/40633348/313196146-1ab572e8-f1eb-4751-af5a-78a5e958b747.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3NjMyNzUsIm5iZiI6MTcxMDc2Mjk3NSwicGF0aCI6Ii80MDYzMzM0OC8zMTMxOTYxNDYtMWFiNTcyZTgtZjFlYi00NzUxLWFmNWEtNzhhNWU5NThiNzQ3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDExNTYxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkMDZhYWZjYjc2M2Q0YzEwMmZiMGIxYzA2NGIyNTRmYmU0ZWFjZGQ4ZWRjZTA4NDkyOTcxMjU3ODk2MDUxZGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7vBE94d9OKGrmeYim9IQ4byB49_TzZ39yAdkJHoK5uc)

Ok done

> 
> ![image](https://private-user-images.githubusercontent.com/40633348/313196324-77e42140-54f0-46df-bcdf-a21933b191f2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3NjMyNzUsIm5iZiI6MTcxMDc2Mjk3NSwicGF0aCI6Ii80MDYzMzM0OC8zMTMxOTYzMjQtNzdlNDIxNDAtNTRmMC00NmRmLWJjZGYtYTIxOTMzYjE5MWYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDExNTYxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc1MTQ3OGFhOWM3M2YyOTZmMDc1N2NmMGVkNTA5ZjZkZmNkNmNmZmU2ZTVkOGM2YjE4OWExMzgzNmUwNWI2NzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.T-OaukV0lQstvstPDurnd3eZUPvraAJ1tLuN-0fBQXM)
> 

This Section you are referring to is interesting. I do not think commit has a very concrete place in the instruction flow. In many implementations, it might happen at the same time as result.

I think it should be possible for implementations to chose to have commit earlier than register. Especially as in split scenario, a register transaction is not be initiated if the operand is not available. Then there is time for an exception or so which would result in the transaction being killed, before its operands are provided.

> ![image](https://private-user-images.githubusercontent.com/40633348/313196609-26790791-07de-467e-8cbc-19754825310f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3NjMyNzUsIm5iZiI6MTcxMDc2Mjk3NSwicGF0aCI6Ii80MDYzMzM0OC8zMTMxOTY2MDktMjY3OTA3OTEtMDdkZS00NjdlLThjYmMtMTk3NTQ4MjUzMTBmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDExNTYxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIzNTY0ODlhNzY1MGViNDEwMDQ2N2Y3NzZhMmViNWI3MDQyMTA1N2IzZTQwN2YzYzA0MGUzZTM2N2JjMmE2MTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qH6E6w9G5bncQrzdyYrnAHvbTRRGRnrb8Gx1AUpYxZs)
> 

True. I updated.

> ![image](https://private-user-images.githubusercontent.com/40633348/313196747-4aa0762c-eeff-491b-9974-11ddf181858c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3NjMyNzUsIm5iZiI6MTcxMDc2Mjk3NSwicGF0aCI6Ii80MDYzMzM0OC8zMTMxOTY3NDctNGFhMDc2MmMtZWVmZi00OTFiLTk5NzQtMTFkZGYxODE4NThjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDExNTYxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAyNzY5NjhmNjg2YTdjOTYxODFhYTBmMDI3NmQyZGQ4ZTUxODc5YTcxZDFkNjUwMmExNjJiNGIyZDUyYzUyYTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OexdSB_WgOoYJrOvvSPLoIomr1i-HjIJL49-cecfj3Q)
> 

Issue should be the only limitation. If issue happens, but result doesnt for a while, a commit with kill should be possible.

> ![image](https://private-user-images.githubusercontent.com/40633348/313197032-ad7f18fe-5047-4c72-9a70-95990e1e5f32.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3NjMyNzUsIm5iZiI6MTcxMDc2Mjk3NSwicGF0aCI6Ii80MDYzMzM0OC8zMTMxOTcwMzItYWQ3ZjE4ZmUtNTA0Ny00YzcyLTlhNzAtOTU5OTBlMWU1ZjMyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDExNTYxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZhMzUzMTczZThkOTZhYjUyNzQyMTJjMGYyZWNkYzZiMjNlODk4YWFmYjkwZTAyMzM2MWJjMzIyMjcxMTc2ZjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.kX1vIz-sidr6tbnGhesK5Tob-RXY6tR5e7DI2TrjEHE)

ok done

christian-herber-nxp, 2024-03-18:
additional note: I did also share the interface sequence, so now commit is before register. I do think however, that this section is a bit confusing, as it refers to an instruction flow, and that may clearly be different (e.g. commit after an eventual memory transaction).

closed 2024-04-15 by christian-herber-nxp