openhwfoundation/core-v-xif #227: Use of SystemVerilog Interface should be strongly discouraged

MikeOpenHWGroup, 2024-06-12:
### Is there an existing CV-X-IF task for this?

- [X] I have searched the existing task issues

### Task Description

This repo hosts an example SystemVerilog Interface of the CV-X-IF in [src/core_v_xif.sv](https://github.com/openhwgroup/core-v-xif/blob/f08d951d52e3705b54767bbf13dbb4748383fe49/src/core_v_xif.sv#L1).  It should be made clear that this is a _simulation only_ example and should not be used in any CORE-V RTL IP.

The CV32E40P [coding style guidelines](https://github.com/openhwgroup/cv32e40p/blob/782283af77ee3b9cec8f98c9e10a0582537a2ff0/README.md?plain=1#L63), which are essentially the [lowRISC coding style guidelines](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md), indicate that SystemVerilog Interfaces are [problematic constructs](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#problematic-language-features-and-constructs) and their use is discouraged.

### Description of Done

[core_v_xif.sv](https://github.com/openhwgroup/core-v-xif/blob/f08d951d52e3705b54767bbf13dbb4748383fe49/src/core_v_xif.sv#L1) is a useful example, so it is proposed that we simply add the following to the comment header of that file:
```
// EXAMPLE ONLY
// The lowRISC coding style guidelines (https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md)
// indicate that SystemVerilog Interfaces are problematic constructs and their use in CORE-V RTL IP is strongly discouraged.
```

Silabs-ArjanB, 2024-06-13:
I strongly disagree with this. Use of SystemVerilog interfaces should actually be encouraged for certain uses (including this one). Removing myself from this issue.

christian-herber-nxp, 2024-06-13:
i think cv-x-if should neither encourage or discourage particular coding styles are choices.

MikeOpenHWGroup, 2024-06-13:
Hi @Silabs-ArjanB, can you provide a specific examples of where/how SV Interfaces should be used in RTL?  This issue was motivated by a recent discussion within the CVA6 team in which both myself and @Jbalkind

Silabs-ArjanB, 2024-06-13:
Hi @MikeOpenHWGroup 

>> can you provide a specific example of where/how SV Interfaces can be safely used in RTL? 

https://github.com/openhwgroup/cv32e40x/blob/master/rtl/cv32e40x_core.sv

>> I will reiterate that the OpenHW Cores Task Group adopted the [lowRISC coding style guidelines]

We treat the lowRISC coding style guidelines as exactly that, i.e. guidelines. Mostly these are good guidelines, but we don't take them too literally. They were only adopted because nobody wanted to spend time/effort in writing our own guidelines and/or opening up internal guidelines.

davideschiavone, 2024-06-19:
I agree that the use of the SV interfaces should neither be encouraged nor discouraged and that the lowRISC coding style is a guideline, but not a strict rule.

There are many reasons why SV interfaces are the preferred choice, X-IF would be  one example (although can be done with structures as well)

Jbalkind, 2024-06-19:
I have to say that I disagree with the suggestion we should be embracing interfaces. When this was discussed in the CVA6 meeting, there were multiple responses from OpenHW members saying that they had designs which were broken by the use of interfaces. One member has a chip which does not function correctly because Genus didn't handle interfaces correctly. We should not be distributing code which has known problems when used with well known tools.

I like interfaces and I wish that they were better supported but I do not think we should be including them in our codebases until we are past the point that they are a common concern. If we are giving types using interfaces then we should also be giving associated structs as examples to avoid what has already happened (contributors wasting their time to use interfaces when their use shouldn't be merged).