openhwfoundation/core-v-xif #231: Clarification needed: Is "x_compressed_req.instr[1:0] == 2'b 11" legal processor-behavior?

silabs-robin, 2025-07-11:
There is one ambiguous sentence in the spec that should be clarified.

What the spec says:
`A coprocessor may only accept valid 16-bit instructions, i.e. bits [1:0] must not be binary 11.`.

Two possible interpretations:
1. For a coprocessor to accept a 16-bit instruction, then bits [1:0] must not be binary 11.
2. When a processor offloads a 16-bit instruction, then bits [1:0] must never be binary 11.

--------

It seems the [core-v-verif xif asserts](https://github.com/openhwgroup/core-v-verif/blob/4aff1da1e47c281401b3ed7ea388e1ad9470204b/lib/uvm_agents/uvma_cvxif/src/uvma_cvxif_assert.sv#L30) chose the latter interpretation.

I think the former interpretation seems slightly more reasonable. _(So a processor implementation can choose whether it wants to just offload everything, or it can be particular if it wants to.)_

In either case, I think the phrasing is ambiguous and needs to be cleared up in the spec.

silabs-robin, 2025-07-11:
@christian-herber-nxp, @davideschiavone, @MikeOpenHWGroup, are any of you still active on this repo?
_(If not then sorry for the ping. There is an "Unsubscribe" button on the right.)_

christian-herber-nxp, 2025-07-11:
as the sentence specifies the acceptance behavior of the coprocessor, I believe attempting to offload a compressed instruction with LSBs 11 is legal for a processor.

silabs-robin, 2025-07-11:
Hi @christian-herber-nxp.
Your reasoning is the same that I followed.

I made this PR to disambiguate it: https://github.com/openhwgroup/core-v-xif/pull/232

silabs-robin, 2025-07-14:
There is already a note in the spec on this: `It is not required for a CPU to ensure that the offloaded instruction is a valid 16-bit encoding.`